UedaTakeyuki/mh-z19 #1: Additional functionality: Calibration, automatic baseline correction and tempterature

DFyson, 2019-01-18:
There are several other features available on the sensors as per the datasheets for [MH-Z19](https://www.winsen-sensor.com/d/files/PDF/Infrared%20Gas%20Sensor/NDIR%20CO2%20SENSOR/MH-Z19%20CO2%20Ver1.0.pdf), [MH-Z19B](https://www.winsen-sensor.com/d/files/infrared-gas-sensor/mh-z19b-co2-ver1_0.pdf).

I have an MH-Z19B and noticed it has automatic baseline correction which resets the reading to 400ppm. This has been brought up in other [projects](https://github.com/kizniche/Mycodo/issues/545)

Support for calibration (zero point and span)

Although not documented, apparently can read temperature [[1]](https://www.home-assistant.io/components/sensor.mhz19/)[[2]](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-MH-Z19B-CO2-Sensor)

UedaTakeyuki, 2019-01-18:
Hi Dyson,
Thank you for your request.

I didn't support calibration feature due to lack of the appropriate apparatus for test the feature.
But I can add these soon (without test).

Also, ABC feature for MH-Z19B seems to be cause of  terrible erroneous measure because CO2 concentration tend to rise about 1500 ppm in the closing room in my experience. 
I'll soon add the feature for (on/off) of ABC.

About temperature, I think this is sensor internal feature for NDIR because the measurement of NDIR is affected by this.

UedaTakeyuki, 2019-01-18:
Implemented as follows:

- https://github.com/UedaTakeyuki/mh-z19/wiki/module
- https://pypi.org/project/mh-z19/

closed 2019-01-18 by UedaTakeyuki