UedaTakeyuki/mh-z19 #5: Add option to read temperature

rzarajczyk, 2019-02-23:
According to this webpage https://revspace.nl/MHZ19 it should be possible (although not documented in official datasheet) to read the temperature along with CO2 level. Would it be possible to include this information in the result?

UedaTakeyuki, 2019-02-23:
Hi rzarajczyk,

Thank you for your request. It seems to be nice idea!
I'll try to add it soon.

UedaTakeyuki, 2019-02-24:
Hi rzarajczyk,

I've updated [mh_z19.py](https://github.com/UedaTakeyuki/mh-z19/blob/master/mh_z19.py) to return all of [command 0x86](https://revspace.nl/MHZ19#Command_0x86_.28read_concentration.29) feedback as follows:

```bash:
sudo python3 mh_z19.py --all
{"UhUl": 10523, "temperature": 17, "TT": 57, "co2": 539, "SS": 232}

sudo python -m mh_z19 --all
{"SS": 232, "UhUl": 10523, "TT": 57, "co2": 551, "temperature": 17}
```

How do you feel about it? Thank you for your feedback!

rzarajczyk, 2019-02-24:
I'm really impressed with the implementation time! The implementation itself seems to be correct - it does provide results according to the spec. 

However I'm disappointed with the accuracy of MH-Z19 thermometer - in my setup (RPI 1B + MH-Z19 [TBH I don't know which version]) it shows 29-30 °C in the room where actual temperature is 22 °C. So either the the thermometer is inaccurate, or the revspace article was wrong.

Anyway thank you for immediate response! I really appreciate your code!

closed 2019-02-24 by rzarajczyk

UedaTakeyuki, 2019-02-24:
Hi rzarajczyk,

IMHO, the TT value might be related temperature, but this may NOT be a **ambient temperature**, but also heated temperature for measuring CO2.

The principle of CO2 concentration detection is called NDIR(Nondispersive Infrared ). As a rule of thum, this calibrate the temperature near the thermal infrared heater which is dropping down due to absorption by CO2. Detail explanation of NDIR is available [here](https://www.analog.com/en/analog-dialogue/articles/complete-gas-sensor-circuit-using-nondispersive-infrared.html). Figure 1. of the document might be informative to get intuitive understanding.

I think **TT** value must be related the NIDR calibration value which is heated a bit.

For calibrate room temperature, I think [DHT22](https://monitorserviceatelierueda.blogspot.com/2018/10/how-to-get-time-series-chart-of-room.html) might be one of the good sensor module with good accurate and inexpensiveness.

By the way, please feel free to ask me anything. I also really appreciate if I can help you!