UedaTakeyuki/mh-z19 #6: Pip package should depend on setuptools and wheel

sw, 2019-03-26:
`pip install mh-z19` fails if the packages `setuptools` and `wheel` are not installed.

UedaTakeyuki, 2019-03-27:
Hi SW,

Thank you for reporting; and sorry for bother you.
I'll fix it and release updated version; by the time, please install **mh-z19** as follows:

```bash:
pip install setuptools wheel mh-z19
```

UedaTakeyuki, 2019-03-27:
Hi SW,

I've tried to make package with adding ** setuptools** and  **wheel** on **install_requires** as version 0.3.8 on [testpi](https://test.pypi.org/project/mh-z19/).
Then removed setuptools and wheel package, and tried to install it as follows:
```bash:
pip install --extra-index-url https://testpypi.python.org/pypi mh-z19
Collecting mh-z19
  Using cached https://files.pythonhosted.org/packages/9d/85/12432906d4b15ab18bb9c284ffbdca529e8eb6d4a0a0616efba443a1b0f3/mh_z19-0.3.8.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ImportError: No module named setuptools
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-S0hGki/mh-z19/
pi@raspberrypi:~ $ pwd
```

The module **setuptools** is used by pip to get egg_info in which is defined **install_requires**.
It might be no way to install **setuptools** by install script of **mh_z19**.

So, please install **setuptools** and **wheel** before install **mh_z19** as follows:

```
pip install setuptools wheel mh-z19
```

Thank you.

UedaTakeyuki, 2019-03-28:
I'm closing this.

closed 2019-03-28 by UedaTakeyuki