UedaTakeyuki/mh-z19 #12: Use different serial port

Actpohomoc, 2019-07-27:
I have used already built-in serial connection for RFID reader. Is there way to use USB serial device to get information from CO2 sensor?
Or implement software serial like we have it on Arduino.

UedaTakeyuki, 2019-07-27:
Hi Actpohomoc,

Thank you for your interesting and helpful suggestive question!
I.M.H.O. in the aspect of the software, the main difference in RPi UART  and USB serial is just a tty device name (and some other parameters). What tty device do you see for the USB serial? Also, do you see the appropriate serial parameter?

TBR-BRD, 2019-07-29:
Hello, i have the same problem. ls -l /dev/serial/by-id => lrwxrwxrwx 1 root root 13 Jul 29 18:56 usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -> ../../ttyUSB0
Which config must be edit to use this USB UART module to read the data?
Thank you

UedaTakeyuki, 2019-08-01:
HI TBR-BRD

How was /dev/ttyUSB0?
How to find USB serial is:

1. Without USB serial, look ls /dev
2. Connect USB serial, then look ls /dev again, find an added one must be the device file of USB serial.

UedaTakeyuki, 2019-08-11:
Hi TBR-BRD,

I've added **--serial_device** option for specifying a specific serial device.
Please try it with the latest version of the library as follows:

```
sudo python -m mh_z19 --serial_device /dev/ttyUSB0
```

Thanks, and sorry for my late reaction due to my summer vacation. Now, I've returned back home.

TBR-BRD, 2019-08-12:
Hi, thank you for your work. I will test it later this week. Thanks

UedaTakeyuki, 2019-08-22:
no feedback.

closed 2019-08-22 by UedaTakeyuki