UedaTakeyuki/mh-z19 #13: Question about enabling serial port

joshuajon, 2019-08-20:
I was attempting to follow the instructions at https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-Enable-Serial-Port-hardware-on-the-Raspberry-Pi but ran into issues with permissions on the /dev/ttyS0 port on (rpi3).  I added the enable_uart=1 in /boot/config.txt and rebooted, the device was set with permissions 620.  I could chmod them but it would reset on boot.  

The only way I found to fix this issue was to use the raspi-config instructions instead.

UedaTakeyuki, 2019-08-21:
Hi, joshuajon !

Thank you for asking, but I'm afraid I haven't understood what you are asking.
If you don't mind, would you please tell me more detail about what you are stuck on. 
Thank you for your help to understand the problem you are in trouble.

joshuajon, 2019-08-21:
When I read the instructions at https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-Enable-Serial-Port-hardware-on-the-Raspberry-Pi it seemed like there were two options:

1) add enable_uart=1 to /boot/config.txt and reboot
2) use raspi-config to enable the serial interface

After performing step 1, the permissions are: 
crw--w---- 1 root tty 4, 64 Aug 20 08:43 /dev/ttyS0

Only after performing step 2 were the permissions correctly set to:
crw-rw-rw- 1 root dialout 4, 64 Aug 20 08:44 /dev/ttyS0

That's all.  I believe I've worked through the issue, I was just trying to understand if this was the correct / expected behavior or if I misunderstood the instructions.

UedaTakeyuki, 2019-08-21:
Hi joshuajon,

Thank you for giving me detail and sorry to bother you.
I didn't understand the difference in the permissions you mentioned and it affects when using this library with **dialout** group. I've really appreciated your information. Thank you.

UedaTakeyuki, 2019-08-21:
I've updated the [page](https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-Enable-Serial-Port-hardware-on-the-Raspberry-Pi).
Thank you for your informative information.

UedaTakeyuki, 2019-08-22:
no feedback.

closed 2019-08-22 by UedaTakeyuki

Philipp91, 2023-06-10:
After debugging an issue for a while today where `mh_z19` couldn't read from `/dev/ttyS0`, let me post this info to anyone who finds this thread in the future (as it was the first place I looked):

* After a fresh boot, your `ls -l /dev/ttyS0` should show `crw-rw---- 1 root dialout`.
* If it does not, make sure the serial console is disabled in `raspi-config`, as per the instructions in this repository.
* If you encounter `crw--w---- 1 root tty` at a later time again, note that the `mh_z19` library itself is capable of messing this up, as it closes and later reopens a console on the serial port, and apparently it (sometimes) opens even if it was never open before. **If you're using the `mh_z19` library only/mostly in non-root mode, be sure to always pass `--serial_console_untouched` in all your function calls or CLI invocations.**