UedaTakeyuki/mh-z19 #14: Not getting data

KrunchMuffin, 2019-11-14:
Hi,

I have the mh-z19b. I have installed the library but I am not getting data.
[image]

I also tried in a python file with

```
import mh_z19
mh_z19.read()
```
I get `none` back with that.

KrunchMuffin, 2019-11-16:
Not sure what I did, but I got it working.

closed 2019-11-16 by KrunchMuffin

pepsifan92, 2019-11-17:
I had the same `null` - results after install and connected sensor.
Until I **switched the 5V power supply of the sensor off and on again**. After a few seconds, I got expected co2 results.

KrunchMuffin, 2019-11-17:
Ya, I think that's what I did.

UedaTakeyuki, 2019-11-18:
Hi, KrunchMuffin,

I'm glad to hear you can get data!
Sorry for my late reply.

flaviusone, 2020-04-16:
Also getting this after around 3 days of usage (reading every 10 seconds). Restarting the power supply is not a solution for a long time monitoring system. Can we do something from software @UedaTakeyuki?

UedaTakeyuki, 2020-04-16:
Hi, flaviusone

I'm afraid I have no idea. If restarting works solution, add the logic of automatic restarting can be work when null returns continuously.

kpmkpm, 2020-07-30:
I have the same issue. Gives null values or 9 with --all flags.

5v Off and On does not work. Any ideas? Broken sensor?

UedaTakeyuki, 2020-07-30:
Hi kpmkpm,

I think this is hardware matter. The most possible cause might be that the cable connection is loose.

Alexs784, 2020-10-03:
Hey there! I am having this exact same issue on my Raspberry 3 :/ It used to work, but it doesn't anymore... It stopped working when my old power adapter broke. What I did is order a new one wich supplies 3A if I remember correctly, ordered a new sensor as well (I thought the old one was _broken/burnt_) and changed jumpers but I still get `null` (or `9` when using the `--all` option).
Don't know what else to try... Any ideas?

kpmkpm, 2020-10-04:
> Hey there! I am having this exact same issue on my Raspberry 3 :/ It used to work, but it doesn't anymore... It stopped working when my old power adapter broke. What I did is order a new one wich [...]

My problem was loose soldering. After resoldering the contacts it started to work.

pepsifan92, 2020-10-06:
You say it's not working since the power adapter changed. I observed, especially with on the raspberry pi (Version >=3), that the supply should output 5,5V, not only 5V, to keep the voltage on the pi always above 5V. The Pi consumes a lot of current in peaks and the voltage drops significant on the board of the pi, because of:
- too long cables
- too thin cables
- too low voltage

Your **mh-z19** could be affected by this issue and the voltage drops there, too. 

Solutions are: 
- short, thick cables
- a capacitor directly connected to 5V & GND pins on the Pi (and your mh-z19) to prevent voltage drops
- Enhance the voltage

I would try the first two things first, maybe it's enougth to prevent voltage drops. 

_(And yes, 5V is the standard voltage for USB. It's a fail on the Raspberry to require 5,5V I think. A 2A Power Adaper should be enough, a pi consumes less than 1A most of the time)_