UedaTakeyuki/mh-z19 #16: CO2 measurement from script

Janusmartin, 2019-12-18:
Dear Sir

I am trying to read out CO2 values by running a python script with no success.
I have successfully been able to read CO2 values using this command from terminal:
sudo python -m mh_z19
And by opening python within the terminal and running the two lines:
import mh_z19
mh_z19.read()

If I make a script with the two lines of code mentioned and run it in terminal ore IDE, however, I am not able to read out CO2 values by get the following error message:
[image]

Could you help me figure out what I am doing wrong?

nincube8, 2019-12-18:
Permission denied means you need to run as super user.
./autostart.sh will set up the 5 minute timer to send data to server.
./autostart.sh --status will tell you if yout monitor service is running
through the terminal.
./autostart.sh --on means you will setup the monitor service to communicate
with the mh_z19.service
If you get [Errno 13] permission denied it means you need super user
access.
Run as sudo

nincube8, 2019-12-18:
CO2.py has an indented block and must befixed to run correctly.

Janusmartin, 2019-12-19:
Dear nincube8

Thank you for your great help, it works :-)

closed 2019-12-19 by Janusmartin