UedaTakeyuki/mh-z19 #17: Usage in Python 3 without sudo

ralphbe91, 2020-05-14:
Is it possible to use this library with Python 3 without sudo? 

When I use "import mh_z19" and "mh_z19.read()" in my program it doesn't work. Although I have activated the serial port (just as you and others decribed ist), I get the following error: PermissionError: [Errno 13] Permission denied: '/dev/serial0'. The sensor is working, because when I use "$ sudo python -m mh_z19" I get the desired result. 

I am a beginner in Python, that has already taken up a lot of my time.

UedaTakeyuki, 2020-05-14:
Hi ralphbe91,

I'm afraid that the sudo might be necessary due to being required to read the serial port on RPi. I think your program calling mh_z19.read() also need the sudo to call without permission error.

ralphbe91, 2020-05-15:
Hi UedaTakeyuki . Oh, too bad. Thanks for your quick response and your effort. If anyone has a solution, I'd love to hear it.

UedaTakeyuki, 2020-05-16:
Hi ralphbe91,

After considered deeply, I've got an idea to improve this library to use without root permission.
I'll try it and tell you the result soon.

By the way, would you please tell me the result of ```ls -la /dev/serial0```?

UedaTakeyuki, 2020-05-16:
Hi ralphbe91,

I've updated this library for non-sudo support. Please refer [this](https://github.com/UedaTakeyuki/mh-z19#how-to-use-without-root-permission) and try it. Thank you!

ralphbe91, 2020-05-17:
Hi, thanks for your effort! But I have still the following problem:
```
pi@raspberrypi:~ $ sudo python3 -m mh_z19 
{"co2": 565}
```
Great!

```
pi@raspberrypi:~ $ python3 -m mh_z19 --serial_console_untouched
usage: __main__.py [-h]
                   [--serial_device SERIAL_DEVICE | --version | --all | --abc_on | --abc_off]
                   [--span_point_calibration SPAN_POINT_CALIBRATION]
                   [--zero_point_calibration] [--detection_range_5000]
                   [--detection_range_2000]
__main__.py: error: unrecognized arguments: --serial_console_untouched
```
Not that great...

I get the same error in my program when I use `mh_z19.read(serial_console_untouched=True)`: TypeError: read() got an unexpected keyword argument 'serial_console_untouched'

Is there a problem with Python 3?

Thanks!

UedaTakeyuki, 2020-05-18:
Looks like the library you are using hasn't been updated. Would you please make sure the version?

```
pi@raspberrypi:~ $ python3 -m mh_z19 --version
0.5.1
```

The expected version is 0.5.1. Thank you!

ralphbe91, 2020-05-19:
Thats it!

After hours I got it working, thanks to your help. Keep up the good work!

closed 2020-05-19 by ralphbe91