UedaTakeyuki/mh-z19 #19: Add additional value for detection range

WO15, 2020-07-06:
Currently there are 2 detection ranges you can select/set 2000ppm and 5000ppm see 
https://github.com/UedaTakeyuki/mh-z19/wiki/CALIBRATION-&-detection-range
in the datasheet v1.5 in chapter 5 on page 5 there are 3 ranges indicated 2000, 5000 and 10.000.
Would it be possible to add the 10.000ppm range as a "detection_range_10000()" command to the code?
on page 8 of the datasheet the details are given.
Set 0~10000ppm detection range, send command: FF 01 99 00 00 00 27 10 2F

UedaTakeyuki, 2020-07-06:
Hi WO15,

Thank you for your kind suggestion. I'll implement it soon.

UedaTakeyuki, 2020-07-07:
Hi WO15,

I've added the **--detection_range_10000** option and updated both github and pypi.

WO15, 2020-07-07:
Upgraded the python package on a Raspberry and tested the 2k, 5k and 10k range setting. Placed the sensor connected with long wires in a drinking glass and exhaled on it for approx 1 minute. Exhaled air contains approx 4% (40k ppm CO2). Did this 3 times for all 3 ranges, the read out values maxed out at the max value for each range set. So the --detection_range_xxxx option for all 3 ranges seems to work OK.

closed 2020-07-08 by WO15