UedaTakeyuki/mh-z19 #21: --serial_device not working

idegre, 2020-08-26:
PLEASE DON'T FORGET TO "STAR" THIS REPOSITORY :)

If you rather keep your project secret, feel free
to email the author at `<ueda@latelierdueda.com>`; any
private correspondence is treated as confidential.

When reporting a bug please include the following:

**Describe the bug**
when attempting to use the --serial_device parameter it always returns an error.
sorry I don't have the exact error as I changed the code on my pi and didn't get any screenshots

**To Reproduce**
1. call `sudo python3 -m mh_z19 --serial_device /dev/ttyUSB0`
2. see `set_serialdevice(args.serial_device)` is undefined error

**Expected behavior**
the command should work

**Screenshots**
If applicable, add screenshots to help explain your problem.

UedaTakeyuki, 2020-08-26:
Hi idegre,

I really appreciated your repport.
The repported issue is reproduced in my side.
I'll soon fix it.

UedaTakeyuki, 2020-08-27:
Hi, idegre!

I've fixed it, now it works as follows:

```bash:
pi@raspberrypi:~ $ sudo python -m mh_z19 --serial_device /dev/serial0
{"co2": 535}
pi@raspberrypi:~ $ sudo python -m mh_z19 --version
0.6.3
```

Please update your mh_z19 module and tell me the result, thx!

UedaTakeyuki, 2020-08-31:
No feedback.

closed 2020-08-31 by UedaTakeyuki