UedaTakeyuki/mh-z19 #22: Got no valid response

monojk, 2020-08-29:
On my Raspberry 3B+ I get 
**None** from `mh_z19.read()` and **9** from `mhz_19.read_all()`
I am using Python3.6
The serial is `/dev/serial1 -> ttyAMA0`
The mh_z19 is connected RxD <-> TxD  and TxD <-> RxD (physical pins 8 and 10)

There is also another module (SDS011) connected to USB.

Any tip for debugging?

UedaTakeyuki, 2020-08-29:
Hi, monojk!

I'm slightly confused about how to connected mh_z19 with RPi.
So, would you please tell me more detail on how to connect the mh_z19 sensor and RPi?
Is the mh_z19 sensor module connected with RPi directory by UART? Or connected by something USB-Serial converter?

If possible, would you attach a photo of how to cabling? Thank you.

monojk, 2020-08-29:
The connection is according to the cabling in your README.md
[image]

UedaTakeyuki, 2020-08-29:
I'm now clear.
The connection seems no problem.

> The serial is /dev/serial1 -> ttyAMA0

Would you please attach here how was "ls -la /dev"? thank you for your kindly cooperation for the debugging!

nincube8, 2020-08-29:
Make sure that you put RxD to TxD, and TxD to RxD on the Pi. Then you need
to make sure your Pi has created the symbolic links in the system.
Reboot, and check you MH-z19 for the Orange light. Make sure I2C is enabled
as well.

monojk, 2020-08-29:
> I'm now clear.
> The connection seems no problem.
> 
> > The serial is /dev/serial1 -> ttyAMA0
> 
> Would you please attach here how was "ls -la /dev"? thank you for your kindly cooperation for the debugging!
```
pi@rp3b:~ $ ls -la /dev
insgesamt 4
drwxr-xr-x  16 root root        3680 Aug 29 16:11 .
drwxr-xr-x  21 root root        4096 Apr  8  2019 ..
crw-r--r--   1 root root     10, 235 Aug 28 10:22 autofs
drwxr-xr-x   2 root root         640 Aug 28 10:22 block
crw-------   1 root root     10, 234 Aug 28 10:22 btrfs-control
drwxr-xr-x   3 root root          60 Jan  1  1970 bus
crw-------   1 root root     10,  63 Aug 28 10:22 cachefiles
drwxr-xr-x   2 root root        2780 Aug 29 16:11 char
crw-------   1 root root      5,   1 Aug 28 10:23 console
crw-------   1 root root     10,  62 Aug 28 10:22 cpu_dma_latency
crw-------   1 root root     10, 203 Aug 28 10:22 cuse
drwxr-xr-x   7 root root         140 Aug 28 10:22 disk
crw-rw----   1 root video    29,   0 Aug 28 10:22 fb0
lrwxrwxrwx   1 root root          13 Nov  3  2016 fd -> /proc/self/fd
crw-rw-rw-   1 root root      1,   7 Aug 28 10:22 full
crw-rw-rw-   1 root root     10, 229 Aug 28 10:23 fuse
crw-rw----   1 root gpio    254,   0 Aug 28 10:22 gpiochip0
crw-rw----   1 root gpio    254,   1 Aug 28 10:22 gpiochip1
crw-rw----   1 root gpio    247,   0 Aug 28 10:22 gpiomem
crw-------   1 root root     10, 183 Aug 28 10:22 hwrng
lrwxrwxrwx   1 root root          25 Nov  3  2016 initctl -> /run/systemd/initctl/fifo
drwxr-xr-x   2 root root          60 Jan  1  1970 input
crw-r--r--   1 root root      1,  11 Aug 28 10:22 kmsg
lrwxrwxrwx   1 root root          28 Nov  3  2016 log -> /run/systemd/journal/dev-log
brw-rw----   1 root disk      7,   0 Aug 28 10:22 loop0
brw-rw----   1 root disk      7,   1 Aug 28 10:22 loop1
brw-rw----   1 root disk      7,   2 Aug 28 10:22 loop2
brw-rw----   1 root disk      7,   3 Aug 28 10:22 loop3
brw-rw----   1 root disk      7,   4 Aug 28 10:22 loop4
brw-rw----   1 root disk      7,   5 Aug 28 10:22 loop5
brw-rw----   1 root disk      7,   6 Aug 28 10:22 loop6
brw-rw----   1 root disk      7,   7 Aug 28 10:22 loop7
crw-rw----   1 root disk     10, 237 Aug 28 10:22 loop-control
drwxr-xr-x   2 root root          60 Aug 28 10:22 mapper
crw-r-----   1 root kmem      1,   1 Aug 28 10:22 mem
crw-------   1 root root     10,  59 Aug 28 10:22 memory_bandwidth
brw-rw----   1 root disk    179,   0 Aug 28 10:22 mmcblk0
brw-rw----   1 root disk    179,   1 Aug 28 10:22 mmcblk0p1
brw-rw----   1 root disk    179,   2 Aug 28 10:22 mmcblk0p2
brw-rw----   1 root disk    179,   5 Aug 28 10:22 mmcblk0p5
brw-rw----   1 root disk    179,   6 Aug 28 10:22 mmcblk0p6
brw-rw----   1 root disk    179,   7 Aug 28 10:22 mmcblk0p7
drwxrwxrwt   2 root root          40 Jan  1  1970 mqueue
drwxr-xr-x   2 root root          60 Aug 28 10:22 net
crw-------   1 root root     10,  61 Aug 28 10:22 network_latency
crw-------   1 root root     10,  60 Aug 28 10:22 network_throughput
crw-rw-rw-   1 root root      1,   3 Aug 28 10:22 null
crw-------   1 root root    108,   0 Aug 28 10:22 ppp
crw-rw-rw-   1 root tty       5,   2 Aug 29 17:23 ptmx
drwxr-xr-x   2 root root           0 Nov  3  2016 pts
brw-rw----   1 root disk      1,   0 Aug 28 10:22 ram0
brw-rw----   1 root disk      1,   1 Aug 28 10:22 ram1
brw-rw----   1 root disk      1,  10 Aug 28 10:22 ram10
brw-rw----   1 root disk      1,  11 Aug 28 10:22 ram11
brw-rw----   1 root disk      1,  12 Aug 28 10:22 ram12
brw-rw----   1 root disk      1,  13 Aug 28 10:22 ram13
brw-rw----   1 root disk      1,  14 Aug 28 10:22 ram14
brw-rw----   1 root disk      1,  15 Aug 28 10:22 ram15
brw-rw----   1 root disk      1,   2 Aug 28 10:22 ram2
brw-rw----   1 root disk      1,   3 Aug 28 10:22 ram3
brw-rw----   1 root disk      1,   4 Aug 28 10:22 ram4
[... 100 lines cut ...]
crw-rw----   1 root tty       7, 130 Aug 28 10:22 vcsa2
crw-rw----   1 root tty       7, 131 Aug 28 10:22 vcsa3
crw-rw----   1 root tty       7, 132 Aug 28 10:22 vcsa4
crw-rw----   1 root tty       7, 133 Aug 28 10:22 vcsa5
crw-rw----   1 root tty       7, 134 Aug 28 10:22 vcsa6
crw-rw----   1 root tty       7, 135 Aug 28 10:22 vcsa7
crw-rw----   1 root video   244,   0 Aug 28 10:22 vcsm
crw-rw-rw-   1 root root     10,  58 Aug 28 10:22 vcsm-cma
crw-rw----   1 root tty       7,  64 Aug 28 10:22 vcsu
crw-rw----   1 root tty       7,  65 Aug 28 10:22 vcsu1
crw-rw----   1 root tty       7,  66 Aug 28 10:22 vcsu2
crw-rw----   1 root tty       7,  67 Aug 28 10:22 vcsu3
crw-rw----   1 root tty       7,  68 Aug 28 10:22 vcsu4
crw-rw----   1 root tty       7,  69 Aug 28 10:22 vcsu5
crw-rw----   1 root tty       7,  70 Aug 28 10:22 vcsu6
crw-rw----   1 root tty       7,  71 Aug 28 10:22 vcsu7
crw-------   1 root root     10, 137 Aug 28 10:22 vhci
crw-rw----+  1 root video    81,   0 Aug 28 10:22 video10
crw-rw----+  1 root video    81,   1 Aug 28 10:22 video11
crw-rw----+  1 root video    81,   2 Aug 28 10:22 video12
crw-------   1 root root     10, 130 Aug 28 10:22 watchdog
crw-------   1 root root    251,   0 Aug 28 10:22 watchdog0
crw-rw-rw-   1 root root      1,   5 Aug 28 10:22 zero
pi@rp3b:~ $
```

monojk, 2020-08-29:
> Make sure that you put RxD to TxD, and TxD to RxD on the Pi. Then you need to make sure your Pi has created the symbolic links in the system. Reboot, and check you MH-z19 for the Orange light. Make sure I2C is enabled as well.
> […](#)

Hm, the orange LED ist off. Should it be on if power is attached and/or flashing during transfer?

UedaTakeyuki, 2020-08-29:
Hi, nincube8!

Thank you for your support. I think your point is very good. Indeed, in the result of "ls -la /dev" there are **no serial0**, it indicates that **UART is not working**.

As mentioned by  nincube8, most doubtful causes are:

- Did you enable the serial of the RPi? please refer [How to Enable Serial Port hardware on the Raspberry Pi](https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-Enable-Serial-Port-hardware-on-the-Raspberry-Pi )

monojk, 2020-08-30:
All, thanks for the hint. I thought I had enabled serial, but it wasn#T.
Now I am a step further:
```
pi@rp3b:~ $ cat mh-z19.py
#!/usr/bin/python -u
import mh_z19
print(mh_z19.read())
print(mh_z19.read_all())
```
and I get the following results. Notice that **read()** still returns **None** and sometimes there is an error:
```
pi@rp3b:~ $ sudo ./mh-z19.py
None
{'TT': 63, 'UhUl': 25856, 'SS': 0, 'temperature': 23, 'co2': 1441}
pi@rp3b:~ $ sudo ./mh-z19.py
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 490, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/mh_z19/__init__.py", line 63, in mh_z19
    s=ser.read(9)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 497, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
None
{'temperature': 23, 'SS': 0, 'TT': 63, 'UhUl': 25856, 'co2': 1441}

```

UedaTakeyuki, 2020-08-30:
It's a weird occurring serial reading error only in the mh_z19.read(), even not occurring it in mh_z19.read_all() which uses the same logic except for command string.

Indeed, on my side, the issue is not reproduced as follows:

```bash:
pi@raspberrypi:~ $ cat mh-z19.py
#!/usr/bin/python -u
import mh_z19
print(mh_z19.read())
print(mh_z19.read_all())
pi@raspberrypi:~ $ sudo ./mh-z19.py 
{'co2': 751}
{'SS': 235, 'UhUl': 10356, 'TT': 61, 'co2': 751, 'temperature': 21}
```

I wonder there can be something instability of the serial connection, like loose physical cabling, unstable power supply, and so on.

nincube8, 2020-08-30:
Not saying this will work, but it's helped me with sensors in the past. Try
using a 5k pull up resistor across the Vcc and TxD, and if that doesn't
work, try putting the resistor across the RxD and Vcc lines.

nincube8, 2020-08-30:
Make sure that the MH-z19 is lighting up, it has a little laser inside it
and if it's getting good power it will light up orange, that will tell you
if the voltage is good or not.

monojk, 2020-08-30:
Hi nincube8,
the 4.7k pull up resistor across the Vcc and TxD helped. Now I am also getting a result from read(). Thanks.

The orange LED still doesn't light up. I measure a Vcc voltage of 4.8V

Thanks also to UedaTakeyuki for help.

closed 2020-08-30 by monojk