UedaTakeyuki/mh-z19 #23: Funcion abc_off()

temycor, 2020-09-09:
Hola a todos. Tengo una duda:

Si escribo en idle python3:

>>> import mh_z19
>>>mh_z19.abc_off()

¿Se desactiva el autocalibrado del sensor?

Muchas gracias por su gran trabajo.

UedaTakeyuki, 2020-09-10:
Hola temycor!

Si usted entiende.
Sin embargo, no lo he probado porque no tengo mh_z19B, pero no he recibido ningún informe de que mh_z19.abc_off () no funcione.
Si la caribración automática no está desactivada ejecutando mh_z19.abc_off (), háganoslo saber.

temycor, 2020-09-10:
Hi, thank you for your quick answer.

Of course, i have yet the sensor mhz19b but i have to buy the raspberry pi soon. When i buy it, i will write you. 

Buy.

UedaTakeyuki, 2020-09-19:
long time no feedback.

closed 2020-09-19 by UedaTakeyuki

temycor, 2021-03-08:
Excuse me, I haven't bought my raspberry pi yet. :(