UedaTakeyuki/mh-z19 #25: Reported co2 level wrong

rimeraz, 2020-11-22:
Hey,

I received my MH-Z19B and connected it to my raspberry.

I ran a `pip install mh-z19` and when I execute `python -m mh_z19 --all` I receive the following output:
`{"SS": 0, "UhUl": 0, "TT": 50, "co2": 5000, "temperature": 10}`

Temperature is reported more or less correctly (compared against thermometer and DHT22). But co2 value can't be correct. Point of measure was outside.

What else can I do for debugging?

rimeraz, 2020-11-22:
Nevermind.

Running

```
python -m mh_z19 --detection_range_5000
python -m mh_z19 --zero_point_calibration
```

solved the issue. Now it returns:
`{"SS": 0, "UhUl": 0, "TT": 50, "co2": 401, "temperature": 10}`

closed 2020-11-22 by rimeraz

UedaTakeyuki, 2020-11-23:
Hi rimeraz,

Thank you for using this library.
According to the datasheet, need a few minutes to preheat time or return the wrong value.

By the way, please don't forget to give Star to this repository if you like this :-) thank you!