UedaTakeyuki/mh-z19 #26: Returns null - Need help!,

OlafKocanda, 2020-12-13:
Hi there,

first of all thank you for your package. I have recently started playing around with a rpi 4 and wanted to connect a mh-z19b with your python package.

I made sure the serials are available:
```
serial0 -> ttyS0
serial1 -> ttyAMA0
```
Enabled i2c
However i2c output is empty

And even made sure uart is enabled (as seen on [another blog ](https://www.circuits.dk/testing-mh-z19-ndir-co2-sensor-module/) )

When I run sudo python -m mh_z19 I (most of the time) get back
```null```

Sometimes, however there is also another message:
[image]

Here is my cabling set up.
[image]

[image]

I should mention, that the  soldered pins at the mh-z19b where bent a bit due to delivery. But I don't think (hope) this is a problem.

Do you have any ideas what else I could test to make it work?

Thanks in advance!

nincube8, 2020-12-13:
You need a 1-5k Ohm resistor between Vcc+ and TxD line.

OlafKocanda, 2020-12-14:
Hi nincube8,

could you explain a bit what you mean by Vcc+ and TxD line?

Do you want me to put a resistor going from the Vin to the Tx on the MH-Z19B?
[image]

nincube8, 2020-12-14:
You should be able to see the unit light up through the white box on the
side. Usually it's the transfer packet clock that needs a resistor (Tx =
Transfer) between the (Vcc 3v) to be able to decode the strings received by
the unit correctly. The (RxD = Receive line) doesn't need the resistor.

nincube8, 2020-12-14:
I believe this unit is a 5 Volt unit and requires the 5 volt rail on your
Device or Microcomputer.

OlafKocanda, 2020-12-14:
just to make sure: The MH-Z19B is connected to the 5v pin on my raspberry.

Shouldn't the light be on if it receives enough power without any further resistors?

nincube8, 2020-12-14:
Yes, that is correct.

OlafKocanda, 2020-12-14:
Thanks!

Well, then I believe there is something else I am missing here to make it work...

DoudouHirofumi, 2021-01-03:
Hi there.
I had the same problem.
Please try to install "i2c-tools".
`sudo apt install i2c-tools`

After installing it, the "null" problem was solved.

mazdafunsunn, 2021-01-04:
I have same problem. will try to solder.
i2c-tools was already installed.
red LED is blinking.

smrt28, 2021-01-05:
I had the same issue. I added:
```
[all]
enable_uart=1
core_freq=250
```
to /boot/config.txt, reboot and /dev/serial0 works.

mazdafunsunn, 2021-01-06:
> I had the same issue. I added:
> 
> ```
> [all]
> enable_uart=1
> core_freq=250
> ```
> 
> to /boot/config.txt, reboot and /dev/serial0 works.

This worked for me as well.

UedaTakeyuki, 2021-01-19:
no feedback

closed 2021-01-19 by UedaTakeyuki

flipper31, 2021-02-03:
I put a 1k Ohm resistor between Tx and Vin still Null then I Tried 3k Ohm but still null every other solution like fix core_freq to 250 are not working aswell. :/ Do you have any ideas to solve this problem ?

mohshbool, 2021-02-08:
> I put a 1k Ohm resistor between Tx and Vin still Null then I Tried 3k Ohm but still null every other solution like fix core_freq to 250 are not working aswell. :/ Do you have any ideas to solve this problem ?

im having the same problem as you man. pi 4b

UedaTakeyuki, 2021-02-09:
How about using PWM interface?

https://github.com/UedaTakeyuki/mh-z19#pwm-support

mohshbool, 2021-02-09:
> How about using PWM interface?
> 
> https://github.com/UedaTakeyuki/mh-z19#pwm-support

i tried that man, i connected the pwm on the sensor to gpio12, gpio26 and gpio 16 (which is the fifth one from the top right)
here's what `gpio readall` shows: 
```
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 1 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 1 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
```
and here's the error im getting 
```
pi@raspberrypi:~ $ python -m mh_z19 --pwm --pwm_gpio 12 --pwm_range 2000
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/.local/lib/python2.7/site-packages/mh_z19/__main__.py", line 101, in <module>
    print mh_z19.read_from_pwm(gpio=args.pwm_gpio, range=args.pwm_range, )
  File "/home/pi/.local/lib/python2.7/site-packages/mh_z19/__init__.py", line 213, in read_from_pwm
    raise GPIO_Edge_Timeout("gpio {} edge timeout".format(gpio))
mh_z19.__init__.GPIO_Edge_Timeout: gpio 12 edge timeout
```

UedaTakeyuki, 2021-02-10:
For PWM, connect with 3 lines: **5V**, **GND**, and **signal line(Fx: gpio12)**.
refer https://github.com/UedaTakeyuki/mh-z19/wiki/PWM-support.#cabling