UedaTakeyuki/mh-z19 #27: untested zero point calibration command seems to be working as expected

richteas75, 2020-12-17:
As requested on https://github.com/UedaTakeyuki/mh-z19/blob/master/pypi/README.md, I'd like to report that the zero point calibration function seems to be working as expected. When called e.g. by

`sudo python -m mh_z19 --zero_point_calibration`

the current CO2 concentration is set to represent 400 ppm CO2. This was checked in the following way:

1. calling
`sudo python -m mh_z19 --all`
showed a reading of more than 800 ppm.
2. calling  
`sudo python -m mh_z19 --zero_point_calibration`
returned a message saying that ZERO point calibration was carried out.
3. immediately thereafter, without changing, calling again: 
`sudo python -m mh_z19 --all`
returned a reading close to 400 ppm.

Tested using a mh_z19b device connected to a Raspberry Pi Zero W, purchased in the fall of 2020.

UedaTakeyuki, 2020-12-17:
Hi richteas75,

I really appreciate your report! Thank you!

closed 2020-12-19 by UedaTakeyuki

albielew, 2021-03-05:
Did the same thing with Pi3 
before calibration it was 1800
after calibration it was 403

UedaTakeyuki, 2021-03-05:
Hi albielew,
Thank you for your report!