UedaTakeyuki/mh-z19 #28: sudo python -m mh_z19 alway returns null

flipper31, 2021-02-02:
I am using MH_Z19B with a Raspberry Pi 3B
I always get null when I use  sudo python -m mh_z19
I activated Serial in raspi-config 
I wired everything correct but I can't find the problem.
When the sensor is not connected it also shows null 
Could you please help me find the problem ?

UedaTakeyuki, 2021-02-03:
https://github.com/UedaTakeyuki/mh-z19/issues/26#issuecomment-744039360

UedaTakeyuki, 2021-02-05:
How about using PWM interface supported since todday's version?

https://github.com/UedaTakeyuki/mh-z19/wiki/PWM-support.

monojk, 2021-02-05:
Hi, 
I used the PWM interface on another mh-z16 running with an Arduino. The returned values are not stable. They change very often. So I am using the serial on my Arduino.

I have the same  problem with may Raspberry Pi as flipper31. I was there using the gas sensor for many months w/o a problem. Since February I have the problem now. I am using Python 3.5
Please help.

UedaTakeyuki, 2021-02-05:
Hi monojk,

refer https://github.com/UedaTakeyuki/mh-z19#in-case-you-cant-get-the-value

monojk, 2021-02-05:
Hi Ueda,
thanks for the pointer. The permissions on /dev/tty0 fixed my problem. Not sure why it occurred. Maybe due to a system update...

flipper31, 2021-02-05:
Hey monojk can you tell me how I can set the permissions on dev/tty0
I thought if I use sudo I will not run into any permission problems.
And I will try the PWM method
Thanks alot for the fast help

monojk, 2021-02-05:
I used the guidance of https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-use-without-root-permission.
-> these commands
`$ ls -la /dev/serial0
lrwxrwxrwx 1 root root 5 Feb  5 14:31 /dev/serial0 -> ttyS0
$ ls -la /dev/serial0
crw--w---- 1 root tty 4, 64  Feb  5 17:11 /dev/ttyS0
$ sudo chmod g+r /dev/ttyS0
$ sudo addgroup pi tty`

monojk, 2021-02-07:
Hi Flipper31, I followed https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-use-without-root-permission.
I additionally ran: `sudo addgroup pi tty`

UedaTakeyuki, 2021-02-11:
long days no feedback

closed 2021-02-11 by UedaTakeyuki