UedaTakeyuki/mh-z19 #29: Raspberry 4 Support

iperniaf, 2021-03-24:
Hi,
I'm using your lib with a rpbi 4 and it needs some fixes.
¿Can I contribute to the community fixing them?

UedaTakeyuki, 2021-03-24:
Would you please tell me which problem and fixing?
I've not heard of any problem with RRi4.

iperniaf, 2021-03-24:
On mh_z19.py script, line 39, you select serial port using strict Rpi model.
Rpi4 model is not listed, so it tries to use ttyAMA0 port, and the correct one is ttyS0

UedaTakeyuki, 2021-03-24:
Weird, line 29 can't be passed because Raspberry Pi OS must have **/dev/serial0** as a symbolic link.
Do you use other OS?

iperniaf, 2021-03-24:
yes, i'm using a raspberry image on a docker container running on balena etcher OS

UedaTakeyuki, 2021-03-24:
I see!

So, the necessary fix might be as following 2.

1. Add RPi4 definition to https://github.com/UedaTakeyuki/getrpimodel
2. Add "4 Model B" to line 31

right?

UedaTakeyuki, 2021-03-24:
> Add RPi4 definition to https://github.com/UedaTakeyuki/getrpimodel

Sorry, this was my missunderstanding, now RPi4 is in getrpimodel.
So, nesessary fix might be only #2.

I welcome your pull request!

iperniaf, 2021-03-24:
I have no permision to push my changes on a new branch

UedaTakeyuki, 2021-03-24:
Not push directory, instead, you can make **pull request**.

iperniaf, 2021-03-24:
done!
I have an extrage behavior. 
I'm running your script with serial_console_untouched=True
Sometimes the scripts runs ok, but other times it returns "None"
[image]
Can you imagine what's happening?

UedaTakeyuki, 2021-03-24:
No idea, sorry.

UedaTakeyuki, 2021-03-25:
Hi iperniaf,

I've fixed it and updated it to version 3.0.2 both github and pypi.
Would you please try it? thank you.

UedaTakeyuki, 2021-03-29:
longtime no feedback

closed 2021-03-29 by UedaTakeyuki