UedaTakeyuki/mh-z19 #32: "null" on screen.

spidermancan, 2021-10-02:
I have the MH-Z19 attached to the ANAVI phat, on a raspberry pi zero w. I installed the python modules and enabled the i2c interface. I found a similar help request, but I tried all of the different codes and nothing happened. 

THis is my wiring:
Tx-Rx
Rx-Tx
vin-3.3v
GND-GND
When I type in`sudo python -m mh_z19` this output is null. 
[image]

UedaTakeyuki, 2021-10-04:
Hi spidermancan,

- try pull up, refer [issue 26](https://github.com/UedaTakeyuki/mh-z19/issues/26#issuecomment-744039360)

or 

- try [PWM](https://github.com/UedaTakeyuki/mh-z19/wiki/PWM-support.)

UedaTakeyuki, 2021-10-07:
No feedback for 5 days.

closed 2021-10-07 by UedaTakeyuki