UedaTakeyuki/mh-z19 #33: function abc_off() work fine but function read() not work

temycor, 2021-10-03:
Hello everyone, I come to comment on a positive and a negative result. I have raspberry pi 3 b +.
I have tested the abc_off () function for 7 days continuously and it works fine. On the other hand, when I print(mh_z19.read_from_pwm ()) everything goes well and I get the correct reading, but when I print (mh_z19.read()) I get an error. I attach an image with the error that I get. I have previously made the configuration included in this [link](https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-Enable-Serial-Port-hardware-on-the-Raspberry-Pi) but I keep getting the error of the attached image. Somebody could help me?

Thank you

[image]

UedaTakeyuki, 2021-10-04:
Hi temycor

How can I read your attached error?

What does the first line ``{'co2': 345.0} mean? You issued a series of mh_z19.read() then the first one was **succeeded** and 2nd, 3rd one was faulted?

temycor, 2021-10-04:
Hi UedaTakeyuki, I have a script in which I read ppm via pwm and serial port (uart). The first line is the result of the pwm read. When reading through the serial port (uart) the error 13 occurs (attached image) and the response is None. I thought the problem was the serial port configuration, but I have configured the serial port as indicated in its github profile but it also gives error. I hope I have clarified your doubts and can help me solve the error.

UedaTakeyuki, 2021-10-04:
I see.

Do you read ``/dev/serial0`` with **root** permission or other appropriate?

temycor, 2021-10-04:
I am running this script in python 3.7.3:

```
import mh_z19

print(mh_z19.read_from_pwm())
print(mh_z19.read())
```

UedaTakeyuki, 2021-10-04:
Did you run the script with **sudo**?
In general, a serial port requires root permission.

``sudo python3 YOUR-SCRIPT``

temycor, 2021-10-04:
This is the result. I think my mhz-19 is broken. I used it with my arduino and when I did the reading through a serial connection, the sensor did not return any value either. The sensor only returned the concentration value to me when I did the reading via PWM.
Is there a way to run my script like sudo outside of the console?

[image]

temycor, 2021-10-04:
How can I configure raspberry pi 3b + to be able to read the concentration value from my mhz-19 sensor via serial connection from my python script?

UedaTakeyuki, 2021-10-04:
refer https://github.com/UedaTakeyuki/mh-z19#in-case-you-cant-get-the-value

UedaTakeyuki, 2021-10-04:
use without sudo https://github.com/UedaTakeyuki/mh-z19/wiki/How-to-use-in-your-program.

temycor, 2021-10-05:
Thank you. Tomorrow I will implement your suggestions.

UedaTakeyuki, 2021-10-10:
No feedback for 5 days.

closed 2021-10-10 by UedaTakeyuki