UedaTakeyuki/mh-z19 #36: No module named mh_z19

DavidBock, 2021-11-09:
Hello, I'm a newbie to python and it's packages so i blindly followed your instructions to install and use your module from https://pypi.org/project/mh-z19/
I'm connected to my Raspberry over SSH.
Taking the pip install route, which is the easiest but python not finding the module (error in title) led me to try to run it with python3 and even reinstalling it with pip3 left me with no error but also with no success.

So, there are two things I noticed while inspecting the instructions:
```
Install:
pip install mh-z19

Use:
sudo python -m mh_z19
```

1. the use of sudo
2. the use of - and _

What I've did now was like:
```
Install:
sudo pip install mh_z19

Use:
sudo python -m mh_z19
```

And it works. 
Seems like it makes no difference if using - or _ for install but the sudo made a difference.
Am I right that something's not proper with that instructions?

UedaTakeyuki, 2021-11-09:
Hi DavidBock,

Yes, you are right. I should fix these mistakes on the READEM file.
Thank you for your notice.

UedaTakeyuki, 2021-11-10:
Hi David,

I've updated the steps pointed. 
https://github.com/UedaTakeyuki/mh-z19/blob/master/README.md

DavidBock, 2021-11-13:
Looks perfectly clear I'd say and is better than I could've done it.
Good job, thank you. 👍

closed 2021-11-13 by DavidBock