UedaTakeyuki/mh-z19 #37: after install pondslider : Error No module named _init__

herk0, 2021-11-16:
PLEASE DON'T FORGET TO "STAR" THIS REPOSITORY :)

If you rather keep your project secret, feel free
to email the author at `<ueda@latelierdueda.com>`; any
private correspondence is treated as confidential.
pi@raspberrypi:~ $ sudo python -m pondslider --config config.save.toml
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/pondslider/__main__.py", line 6, in <module>
    import __init__ as pondslider
ModuleNotFoundError: No module named '__init__'

UedaTakeyuki, 2021-11-16:
How will it be displayed with the following command?  

```bash:
ls /usr/local/lib/python3.9/dist-packages/pondslider
```

herk0, 2021-11-17:
Hello Dr.Takeyuki Ueda

Tank you for your prompt reply.

I tested the command and the result is:__init__.py  __main__.py  __pycache__

I think that the reason is that i can not see anything from the monitor3.udasoft.com.

I am an absolute newcomer regarding RaspberryPi OS, so i thank you for your efforts and time

you raise for me. I`m looking forward to hear something about them again.

kind regards

Hermann Kautz 

 

 
 

Gesendet: Dienstag, 16. November 2021 um 23:57 Uhr

UedaTakeyuki, 2021-11-17:
It seems strange that the __init__.py is existing on the folder ``/usr/local/lib/python3.9/dist-packages/pondslider`` but message is `` No module named 'init'``.

Just confirmation, can I make sure one point about following your report:

```
File "/usr/local/lib/python3.9/dist-packages/pondslider/main.py", line 6, in
import init as pondslider
ModuleNotFoundError: No module named 'init'
```

Were these really **main** and **init**?  
Instead, weren't **__main__** and **__init__**?

herk0, 2021-11-17:
I have copy the command once again:

 

***@***.***:~ $ ls /usr/local/lib/python3.9/dist-packages/pondslider
__init__.py  __main__.py  __pycache__
 

 
and you see the responce is init and main

Gesendet: Mittwoch, 17. November 2021 um 11:28 Uhr

UedaTakeyuki, 2021-11-17:
Sorry but I've no idea why the python interpreter is complaining ``No module named 'init'`` because collect module name must be ``__init__``.

herk0, 2021-11-17:
Thanks for the honest answer. Do you think, it could help put a new operating system to set up and install MH-Z19 again?

Currently i use OS Buster then i woud use the new once Bullseye. Is that OK?

 

 
 

Gesendet: Mittwoch, 17. November 2021 um 12:05 Uhr

UedaTakeyuki, 2021-11-17:
I'm using MH-z19 on Buster well, but I have no experience with Bullseye.
So no idea about using Bullseye but wide range trial might be help to get some **hint**.
Please try it and tell me what you see. Thank you.

UedaTakeyuki, 2021-11-23:
6 days no feedback.

closed 2021-11-23 by UedaTakeyuki