UedaTakeyuki/mh-z19 #38: error: ubyte format requires 0 <= number <= 255

kzehnter, 2021-12-27:
I have a python program which calls mh_z19.read() every few seconds and sometimes (I don't know why it is only sometimes) I get this error:

> Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/mh_z19/__init__.py", line 73, in mh_z19
    if len(s) >= 4 and s[0] == "\xff" and s[1] == "\x86" and checksum(s[1:-1]) == s[-1]:
  File "/usr/local/lib/python2.7/dist-packages/mh_z19/__init__.py", line 231, in checksum
    return struct.pack('B', 0xff - (sum(array) % 0x100) + 1)
error: ubyte format requires 0 <= number <= 255

Steps to reproduce:
   Call mh_z19.read() every 2 seconds and wait.
   In an hour of runtime it happened around 4 times.

UedaTakeyuki, 2021-12-31:
Thank you for your issue report, kzehnter.
I think I should have cared about the case of the value of ``(sum(array) % 0x100)``is zero.

I'll make fix, sorry my late reply.

UedaTakeyuki, 2022-01-01:
Happy new year, kzehnter!

I've fixed and released it as [version 3.0.5](https://pypi.org/project/mh-z19/).

Before fixing, this issue reproduces a few times in a thousand times.
Now, not reproduce.

kzehnter, 2022-01-02:
Happy new year!
Thank you for fixing that so quickly! :)

I just let version 3.0.5 run on my raspberrypi for a few thousand co2 reads and I haven't gotten the error yet so I can confirm it works!

UedaTakeyuki, 2022-01-03:
Hi kzehnter,

Thank you for your so quick confirmation!
I've really appreciated your report giving me the opportunity to fix a bug that has been around for many years.

Would you mind if I will close this issue?

closed 2022-01-03 by kzehnter