UedaTakeyuki/mh-z19 #41: Character Encoding Issue When Installing on Windows

electronsandstuff, 2022-01-31:
**Describe the bug**
Thanks for the library! I have an issue where pip fails to install it on Windows due to character encoding in `README.md` read by `setup.py`. By the way, my use case is that I am developing code on a Windows platform and then pushing it to a Raspberry Pi to take advantage of my existing IDE setup.

Here is a full copy of the pip error:
```
Collecting mh-z19
  Using cached mh_z19-3.1.1.tar.gz (13 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

  error: subprocess-exited-with-error
  
  python setup.py egg_info did not run successfully.
  exit code: 1
  
  [8 lines of output]
  Traceback (most recent call last):
    File "<string>", line 2, in <module>
    File "<pip-setuptools-caller>", line 34, in <module>
    File "C:\Users\Christopher Pierce\AppData\Local\Temp\pip-install-n1zxmenn\mh-z19_8fc0a2b6efaf4a988424c84193cf061c\setup.py", line 4, in <module>
      long_description = f.read()
    File "C:\Users\Christopher Pierce\AppData\Local\Programs\Python\Python39\lib\encodings\cp1252.py", line 23, in decode
      return codecs.charmap_decode(input,self.errors,decoding_table)[0]
  UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 10805: character maps to <undefined>
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

Encountered error while generating package metadata.

See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new venv on a Windows computer
2. Run `pip3 install mh_z19`

**Expected behavior**
The library is installed.

electronsandstuff, 2022-01-31:
Ok, just added a PR that seems to resolve this. For those reading in the future, you do need to force the package to install without the dependency `RPi.GPIO` which cannot be compiled on Windows. However, even without this, having the package does supply nice autocomplete information for those developing on Windows and pushing to a Raspberry Pi.

UedaTakeyuki, 2022-01-31:
Hi Christopher M. Pierce,

Thank you for telling me your interesting need and nice PR for this.
I've merged it and updated it.

UedaTakeyuki, 2022-02-06:
6 days no feedback.

closed 2022-02-06 by UedaTakeyuki