UedaTakeyuki/mh-z19 #44: Output always empty array

DeckardCain2014, 2022-03-02:
Hi,
I've cloned the most recent repo version and connected everything as shown in the Readme. However, it seems no value is received. This is my output and my connections:
```
sudo python -m mh_z19
{}
```
[image]

UedaTakeyuki, 2022-03-02:
Hi DeckardCain2014,

``Allways returning blank array without any error message nor stacktrace``

It indicates that ``the sensor module didn't respond ``. 
Might be some hardware or cabling problem.

DeckardCain2014, 2022-03-03:
Hi,
thanks for getting back to me. Can you tell me what to check furthermore?

UedaTakeyuki, 2022-03-03:
I'm afraid I can't provide suitable advice.
In general, cabling loosenes, electrify, and so on.

DeckardCain2014, 2022-03-03:
Ok thanks

F4Wins, 2022-03-31:
Hi everyone, I had the same problem. In my case I forgot to enable the serial port hardware on my pi. 
Therefore always make sure that in your /boot/config.txt you have the following line:

`enable_uart=1`

or enable your serial port hardware just like described in the wiki (`sudo raspi-config`)

hawtkey, 2022-07-21:
Hi, I had this error and found it to be that the Pi (3b+) uses UART for bluetooth and that can interfere with the co2 module. you can probably figure out a way to use a different channel if you need, but I disabled bluetooth since I don't use it:

 simply add `dtoverlay=disable-bt` to the end of `/boot/config.txt` and reboot.

resting-dove, 2023-01-31:
I had this error as well and after too much time I realised my pin connection was wrong. When i connected RxD <-> TxD and TxD <-> RxD (physical pins 8 and 10) as mentioned in https://github.com/UedaTakeyuki/mh-z19/issues/22#issue-688525199 I received measurements using serialnas well.