UedaTakeyuki/mh-z19 #45: Wrong temperature and wrong co2 value

F4Wins, 2022-04-01:
Hi everybody,

I am currently trying to get some data using the mh-z19. Considering installation and connection, everything works perfectly fine for me. However when I started to track the data (I am writing it into a .csv file), I would always get the same value for temperature and almost the same values for the co2 measuring: 

```
CO2,Temperature,Time
400,3,31/03/2022 21:00:55
400,3,31/03/2022 21:01:10
400,3,31/03/2022 21:01:25
400,3,31/03/2022 21:01:40
400,3,31/03/2022 21:01:55
400,3,31/03/2022 21:02:10
400,3,31/03/2022 21:02:26
400,3,31/03/2022 21:02:41
400,3,31/03/2022 21:02:56
448,3,31/03/2022 21:04:45
446,3,31/03/2022 21:05:00
442,3,31/03/2022 21:05:15
437,3,31/03/2022 21:05:30
438,3,31/03/2022 21:05:46
434,3,31/03/2022 21:06:01
430,3,31/03/2022 21:06:16
432,3,31/03/2022 21:06:31
436,3,31/03/2022 21:06:46
437,3,31/03/2022 21:07:01
439,3,31/03/2022 21:07:17
435,3,31/03/2022 21:07:32
433,3,31/03/2022 21:07:47
438,3,31/03/2022 21:08:02
444,3,31/03/2022 21:08:17
448,3,31/03/2022 21:08:33
448,3,31/03/2022 21:08:48
```

Some information about the environment: the sensor is currently placed inside where the temperature is at 18-20 degrees celsius. 
Additionally I spent a lot of time in the room the sensor is placed in and I didn't open the windows . So I am sure that the co2 concentration is wrong, same goes for the temperature. 

I was thinking that the sensor maybe needs some time to calibrate itself but after running 24 hours and giving me more than 5000 entries with always the same temperature of 3 degrees and nearly always the same co2 value (between 400-500ppm), I don't think its working properly.

Maybe you can help me and know how to fix this Issue or somehow 'calibrate' the sensor again, I am really clueless at the moment. 

I would be very grateful if you can help me.

UedaTakeyuki, 2022-04-02:
Hi F4Wins,

Thank you for using this library!

# Cause
Without the sensor module's internal problem, the most possible cause is the sensor module's ABC(Automatic Baseline Correction) feature. It considers the current CO2 concentration as ``400ppm`` and automatically calibrates by 400ppm every 24 hours. As you know 400ppm is the mean CO2 concentration of outside air. If the ABC is done in high CO2 concentration for example 1500ppm, after that the sensor module considers 1500ppm as ``400ppm``, then the measured value keep near 400ppm.

# How to fix

## 1. set ABC off
ABC feature might be expected to be used in the outer air. To use it in the room, I recommend you to set ABC ``off`` as follows:

```
sudo python -m mh_z19 --abc_off 
```

for more detail, refer https://github.com/UedaTakeyuki/mh-z19/wiki/CALIBRATION-&-detection-range

## 2. calibrate it with the outer air
Calibrate it with outer air (open the window and ventilate enough) and do ``zero point calibration`` as follows

```
sudo python -m mh_z19 --zero_point_calibration
```

This command calibrates the sensor module as the current co2 concentration as 400ppm.

F4Wins, 2022-04-03:
Hi UedaTakeyuki,

thank you for your fast response and for your explanations! I followed your steps and it seems like the sensor is now properly calibrated, the co2 concentration values are more accurate now (it rises when I am in the room and decreases when I open the windows). 

However, the temperature values are still only between 2-4 degrees celsius while my room is at 18-20 degrees celsius, but thats something I can live with (adding 16 will show me the true temperature I guess). 

Thank you for your awesome library!

UedaTakeyuki, 2022-04-04:
Hi F4Wins,

I'm very glad to hear you are now getting more reasonable co2 concentration values.
About "temperature" of the return value of this library with ``--all``, it may have been misleading naming.

First of all, this value is not based on any mh_z19 official document. But based on the return value of the [undocumented command](https://github.com/UedaTakeyuki/mh-z19#undocumented-response-values-of-0x86-command) ``TT`` which is widely believed to be related to ambient temperature as ``TT - 40 = ambient temperature". So, this ``temperature`` value is just calculated so.

I guess this TT value is maybe [sensor module internal temperature](https://github.com/UedaTakeyuki/mh-z19/wiki/Is-the-undocumented-TT-value-on-the-responses-of-command-0x86-really-related-to-the-ambient-temperature%3F) which is necessary to calibrate co2 concentration by NDIR method, so roughly related with ambient temperature with depending on environment and individual modlue.

Sorry for bothering but I'm happy if you would understand that this value is based on these uncertain and not official features, and your ``own correction`` based on your environment and sensor module works well for you. Thank you.

F4Wins, 2022-04-05:
Hi  UedaTakeyuki,,

I can see what the problem here is. Nevertheless with my own correction I am getting reasonable values. After turning the abc feature off the sensor now ran  2-3 days and is still working perfectly fine. Thank your for your support!