UedaTakeyuki/mh-z19 #48: No Data & Serial Port problems

wh1t3chr1s, 2022-10-28:
Hallo, 
I'm very new to all of this, so sorry in advance.

-I installed it via "pip install".
-I've enabled the serial port hardware in raspi-config as well as I2C which I need for the other part of my project.
-I triple-checked the wiring.
-The red led is blinking.
-I used "chmod 777" (just in case) for both /dev/ttyS0 and /dev/AMA0
-I disabled bluetooth just like one person of the other issues stated that it worked for them.
-After I used _mh_z19.read_all()_ once, it reset all the permissions for ttyS0 back to what it was before. Therefore I get this after using it again:
[image]

Help? ^^°

UedaTakeyuki, 2022-10-28:
Hi wh1t3chr1s,

Would you please try ***mh_z19.read_all(true)*** instead of ***mh_z19.read_all()*** and tell me the result, thx!

wh1t3chr1s, 2022-10-28:
I don't know WHY, but  mh_z19.read_all() works now for some reason as well as the (True) version that you just suggested.

wh1t3chr1s, 2022-10-28:
The only thing thats weird is that the temperature doesn't seem to be correct. I have two other ways of getting my temperature, both tell 4 degrees cooler.

UedaTakeyuki, 2022-10-29:
The temperature returned by read_all() is just an expedient value, not a universal truth.
It's just computed as [TT - 40](https://github.com/UedaTakeyuki/mh-z19/blob/master/mh_z19.py#L129) what TT is for the sensor internal and related to ambient temperature.
Under several environments, -40 works as a good supplement, but if it doesn't fit in your environment, you can use it with adding your complement +4.

More detail about [TT](https://github.com/UedaTakeyuki/mh-z19/wiki/Is-the-undocumented-TT-value-on-the-responses-of-command-0x86-really-related-to-the-ambient-temperature%3F).

UedaTakeyuki, 2022-11-03:
5 day no feedback.

closed 2022-11-03 by UedaTakeyuki