UedaTakeyuki/mh-z19 #49: sudo python -m mh_z19 alway returns {}

test597, 2023-03-25:
Hi, I ran the following command as written in the readme
git clone https://github.com/UedaTakeyuki/mh-z19.git
. /setup.sh
but when I do sudo python -m mh_z19, I get No such file or directory: ' /dev/ttyS0't
I enabled serial communication in raspi-config and did sudo python -m mh_z19, but it shows {}.
I also tried PWM but it timed out and could not get the value.

test597, 2023-03-25:
I am using raspberry pi zero w.
I also use USB to communicate with my PC via ssh to control the raspberry pi.

UedaTakeyuki, 2023-03-26:
Hi test597,

The output ``{}`` means that your RPi can read/write a serial interface correctly but with **no feedback**. The possible cause is  **hardware problem**. Fx: the cable problem, the sensor module problem, and so on. Or the serial port is not connected to the sensor module. I'm not familiar with USB OTG, but I'm afraid that doesn't it occupy the serial port.

For PWM, it just uses simple gpio, so a hardware matter is more suspicious.

test597, 2023-03-26:
Okay, I'm on the road now, so I'll try it without USB OTG when I get back.

test597, 2023-03-28:
I did not use USB OTG and it did not work well.
When PWM was used, a very high value of 5100 ppm was displayed.
I used --zero_point_calibration, but the result did not change.

UedaTakeyuki, 2023-03-28:
> I did not use USB OTG and it did not work well.

So, the most doubtable cause is **Hardware**, I recommend you make sure the cabling, sensor module and so.

> When PWM was used, a very high value of 5100 ppm was displayed.

How did you measure with PWM? Did you mean measure with my library as follow?

```
python -m mh_z19 --pwm
```

If so, it's weird. The default range is 5000ppm, so it doesn't indicate over 5000.
Also, calibration by ** --zero_point_calibration** is no mean. It also depends on the Serial connection between **MH_Z19 and RPi**.

test597, 2023-03-28:
Sorry, it was incorrectly CLOSED.

The values measured by running python -m mh_z19 -pwm are running in 5ppm increments like 5100, 5095, and 5115.
Values of 5115 and 5110 are relatively easy to output in a closed environment, and values of 5095 and 5100 in a ventilated environment.

UedaTakeyuki, 2023-03-28:
I'm afraid I have **no idea** to help with this problem.
Please fix your **Hardware problem** yourself. Bay bay.