UedaTakeyuki/mh-z19 #50: Without root permission won't set

cypridox, 2023-11-08:
Thanks for this usefull package and the clear documentation! The installation went well on a RPI 3B and I got sensor readings. I want to use the sensor in Python programs and so I want to get rid of the root permission. Unfortunately this doesn't work for me. 
I used `sudo chmod g+r /dev/ttyAMA0`
using `ls -la /dev/ttyAMA0` this gave me `crw-rw---- 1 root tty 204, 64 Nov  8 20:00 /dev/ttyAMA0`
but when I tried `python3 -m mh_z19 --serial_console_untouched`  I got an error: 

```
`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/serial0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/mh-z19/mh_z19.py", line 66, in read_concentration
    ser = connect_serial()
  File "/home/pi/mh-z19/mh_z19.py", line 62, in connect_serial
    timeout=1.0)
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/serial0: [Errno 13] Permission denied: '/dev/serial0'
{}
```
And then the 'r' on the fifth position is gone and I have to use chmod again. How to solve this?

UedaTakeyuki, 2023-11-14:
The error message reports no permission to read the serial port ``/dev/serial0``.
Could you please check about it?

cypridox, 2023-11-20:
Thanks for your reply. The serial port appeared to be there on /dev/ttyS0. I added the read rights. It required a reboot but now it works without sudo.

UedaTakeyuki, 2023-11-21:
I am glad to hear that it works properly after granting the necessary permissions. Have fun!

closed 2023-11-21 by UedaTakeyuki