UedaTakeyuki/mh-z19 #51: Still return {} after using pull-up resistor and change freq = 250 and enable_uart = 1

EfraimManurung, 2024-04-06:
Hallo Mr. Takeyuki and everyone,

I still have a problem with my MH-Z19C sensor, it is still always treturn {}, can someone help me? is it compatible for the MH-Z19C or only for B? I already read many blogs and github and I cannot solve my problem

I did my work on Raspberry Pi 5, and I also tried it on my 3+ but still not working. I already tried the pull-up resistor both for RX or TX pins. 

[image]
[image]

Thank you so much.

UedaTakeyuki, 2024-04-06:
Hi Efraim,

There is a track record of operation with MH-Z19C.
The MH-Z19 sensor module requires a few minutes **preheat** time. In case not preheated, you may get the value after preheated.

EfraimManurung, 2024-04-06:
Thank you for your response Mr. Takeyuki. I tried again after pre-heating for more than 10 minutes, it is working using ESP32/Arduino Uno but it's not working for the Raspberry Pi. I already followed all your procedures Mr. Takeyuki. Are there other things that I can do?

and what do you mean by a track record of operation with MH-Z19C? 

[image]

UedaTakeyuki, 2024-04-06:
Ah, I mean I don't use this with type C, someone reported that this library worked well with type C.
By the way, the return  value ``{}`` without any error message means that the serial interface is connected normally. This library sent command to the sensor normally, and no response returned.

EfraimManurung, 2024-04-06:
I already tried it using a different serial Dr. Takeyuki, do you have other solutions?

Oh so do you mean this library only intended for MH-Z19B not for series C? 

[image]

UedaTakeyuki, 2024-04-07:
Hi Efraim,

Not only with type B but also with type C.
I heard a lot of reports that this run with MH-Z19C.

What happens if you run it with ``/dev/serial1``? Not with **serial10**.

EfraimManurung, 2024-04-08:
Dear Dr. Takeyuki, thank you for your response

I believe it is working with **serial0** because **serial1** I cannot access it. Do you have another suggestion? I also reinstall the OS of Raspberry Pi 5 but still not working. 
[image]

EfraimManurung, 2024-04-08:
Dear Dr. Takeyuki, I solved it! finally, after I tried using Arduino Uno, I realised there is a different address for Raspberry Pi 5 and you're right. I searched different addresses and matched with /dev/tty/AMA0

[image]

[image]

Thank you so much for your time! I hope this thread will be helpful for someone too.

Efraim.

closed 2024-04-08 by EfraimManurung

EfraimManurung, 2024-04-08:
dear Dr. Takeyuki, 

I have last question, for "sudo python3 -m mh_z19 --all --serial_device /dev/tty
AMA0"

how do I set up the program? is there a parameter for it to setup the serial_dev?

the program like in the example
"import mh_z19

mh_z19.read_all()"

UedaTakeyuki, 2024-04-09:
I'm so glad to hear that your problem was resolved. I appreciate your patient cooperation in investigating and fixing this issue.

Also, thank you for your reporting that there is ``No documentation`` to change serial device. I've added it as [set_serialdevice(device)](https://github.com/UedaTakeyuki/mh-z19/wiki/module#set_serialdevicedevice).

You can use this library in your program as follows:

```
import mh_z19

mh_z19.set_serialdevice("/dev/ttyAMA0")
mh_z19.read_all()
```

EfraimManurung, 2024-04-09:
Dear dr. Takeyuki, it works! thank you very much for your help. I can continue working on my thesis project. 

[image]

UedaTakeyuki, 2024-04-09:
I hope your thesis project is going well. I would be pleased to see your great thesis once it is completed.