UedaTakeyuki/mh-z19 #52: Another `{}` output issue

2ZZ, 2024-04-08:
Hi all,

I have an Mh-z19c connected to a Pi Zero W with TxD and RxD cross connected to pins 8 (GPIO 14) and 10 (GPIO 15) as per the repo readme and I see an orange light flashing inside indicating power.

The problem is the output from mh_z19 always `{}`.

I found the serial output is not passing [this check](https://github.com/UedaTakeyuki/mh-z19/blob/master/pypi/mh_z19/__init__.py#L76):
```
if len(s) >= 4 and s[0] == 0xff and s[1] == 0x86 and ord(checksum(s[1:-1])) == s[-1]:
  return s[2]*256 + s[3]
```

Instead I see values like these examples:
```
ser.read(9): b'BM\x0c\x9c\t\xad\x02\x13\x01' - s[0]=66, s[1]=77, s[-1]=1
ser.read(9): b'(\x00\x00\xe2\x00\x06\x13BM' - s[0]=40, s[1]=0, s[-1]=77
ser.read(9): b'\x0c\x9c\t\xad\x02\x13\x01(\x00' - s[0]=12, s[1]=156, s[-1]=0
ser.read(9): b'\x00\xe2\x00\x06\x13BM\x0c\x9c' - s[0]=0, s[1]=226, s[-1]=156
ser.read(9): b'\t\xad\x02\x13\x01(\x00\x00\xe2' - s[0]=9, s[1]=173, s[-1]=226
ser.read(9): b'BM\x0c\x9c\t\xb2\x02\x15\x01' - s[0]=66, s[1]=77, s[-1]=1
ser.read(9): b'(\x00\x00\xe2\x00\x06\x1aBM' - s[0]=40, s[1]=0, s[-1]=77
ser.read(9): b'\x0c\x9c\t\xb2\x02\x15\x01(\x00' - s[0]=12, s[1]=156, s[-1]=0
ser.read(9): b'\x00\xe2\x00\x06\x1aBM\x0c\x9a' - s[0]=0, s[1]=226, s[-1]=154
ser.read(9): b'\t\xaa\x02\x15\x01(\x00\x00\xe2' - s[0]=9, s[1]=170, s[-1]=226
```

I've followed the advice of all the existing issues I can find:
- Enable serial port via raspi-config
- Ensure `enable_uart=1` is in `/boot/config.txt`
- Set `core_freq=250` in  `/boot/config.txt`
- Set `dtoverlay=disable-bt` in  `/boot/config.txt` to disable Bluetooth
- Add a resistor between TxD and Vin (I used a 2k)

If anyone has any suggestions they would be highly appreicated!

This is the variant of sensor I have:
[image]

UedaTakeyuki, 2024-04-09:
Hi 2ZZ,

The random serial response you've got seems to be that:
- The sensor module is not preheated, in this case, a few minutes after you can get the right value.
- Using the wrong UART
- The sensor module is broken.

Would you please tell me what is happening when you try again with option ``--serial_device /dev/serial1``,  ``--serial_device /dev/ttyAMA0`` or ``--serial_device /dev/ttyS0 ``, also try without ``dtoverlay=disable-bt`` in your``/boot/config.txt``, thank you.

2ZZ, 2024-04-09:
Hi @UedaTakeyuki,

I don't think it's a preheat issue as I've left it powered on for many hours.

Re-enabling BT and rebooting changed the response from those alternative serial devices, the output is below with some extra debug lines added.

## /dev/serial1

BT disabled

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 380, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
termios.error: (5, 'Input/output error')
```

BT enabled

```
ser.read(9): b''
ser.read(9): b''
ser.read(9): b''
ser.read(9): b''
ser.read(9): b''
{}
```

## /dev/ttyAMA0

BT disabled

```
ser.read(9): b'BM\x0c\x8c\t\xac\x02\xce\x01'
ser.read(9): b'*\x00\x00\xe2\x00\x06\xbfBM'
ser.read(9): b'\x0c\x8c\t\xac\x02\xce\x01*\x00'
ser.read(9): b'\x00\xe2\x00\x06\xbfBM\x0c\x8c'
ser.read(9): b'\t\xac\x02\xce\x01*\x00\x00\xe2'
{}
```

BT enabled

```
ser.read(9): b''
ser.read(9): b''
ser.read(9): b''
ser.read(9): b''
ser.read(9): b''
{}
```

## /dev/ttyS0

BT disabled

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 380, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
termios.error: (5, 'Input/output error')
```

BT enabled

```
ser.read(9): b'BM\x0c{\t\xa5\x03/\x01'
ser.read(9): b'*\x00\x00\xe2\x00\x06\tBM'
ser.read(9): b'\x0c{\t\xa5\x03/\x01*\x00'
ser.read(9): b'\x00\xe2\x00\x06\tBM\x0c{'
ser.read(9): b'\t\xa5\x030\x01*\x00\x00\xe2'
{}
```

---

I will put an order in for a replacement sensor to rule that out.

UedaTakeyuki, 2024-04-09:
It seems as follows:

- from BT:  b''
- from Sensor: b'BM\x0c{\t\xa5\x03/\x01'
- from non-existent device：Traceback (most recent call last):

The response from the sensor module seems to repeat following sequence:

- b'BM\x0c{\t\xa5\x03/\x01*\x00\x00\xe2\x00\x06\t'

Do you have any idea about the sequence above?

2ZZ, 2024-04-09:
I really have no idea on that sequence. I have a spare Pi 2 B here so I tried it on there and the issue is the same. 
Once the new sensor arrives I will try again.

UedaTakeyuki, 2024-04-10:
If you don't mind its rough resolution and re-cabling, [PWM](https://github.com/UedaTakeyuki/mh-z19/wiki/PWM-support.) interface is also available.

2ZZ, 2024-04-10:
PWM is working nicely :raised_hands:

```
root@raspberrypi:~# while :; do python -m mh_z19 --pwm; done
{'co2': 480.0}
{'co2': 480.0}
{'co2': 480.0}
{'co2': 480.0}
{'co2': 480.0}
{'co2': 480.0}
{'co2': 475.0}
{'co2': 475.0}
{'co2': 475.0}
{'co2': 470.0}
{'co2': 470.0}
{'co2': 470.0}
{'co2': 470.0}
{'co2': 465.0}
{'co2': 460.0}
{'co2': 460.0}
{'co2': 460.0}
{'co2': 460.0}
{'co2': 460.0}
```

I will still try GPIO again as I already ordered the new sensor.

2ZZ, 2024-04-17:
The new sensor arrived and works straight away.
This is the one I bought: https://www.aliexpress.com/item/1005002038004611

```
python3 -m mh_z19
{"co2": 796}
```

Thanks for the assistance @UedaTakeyuki

closed 2024-04-17 by 2ZZ

GiChoc, 2024-05-16:
Hello @2ZZ , I have exactly the same problem as you. PWM works fine but the readings from UART are as yours. They do not pass [this check](https://github.com/UedaTakeyuki/mh-z19/blob/master/pypi/mh_z19/__init__.py#L76). 

What was the problem eventually with your case? Do you think I have a faulty sensor?

2ZZ, 2024-05-17:
> Hello @2ZZ , I have exactly the same problem as you. PWM works fine but the readings from UART are as yours. They do not pass [this [...]

Possibly, which variety of sensor do you have?
You may find PWM mode works even if UART doesn't.

GiChoc, 2024-06-02:
> Possibly, which variety of sensor do you have? You may find PWM mode works even if UART doesn't.

I've got the MH-Z19B variety. 
I have now given up :(