UedaTakeyuki/mh-z19 #53: Raspberry Pi 5 Support

rewse, 2024-05-04:
mh_z19 3.1.3 doesn’t work with Raspberry Pi 5 because the dependent RPi.GPIO 0.7.1 doesn’t work with it.

It seems RPi.GPIO is no longer maintained because of no response to [the ticket](https://sourceforge.net/p/raspberry-gpio-python/tickets/214/) to support RP5. [The official RP doc](https://www.raspberrypi.com/documentation/computers/os.html#gpio-in-python) is using gpiozero to access GPIO.

```
$ python3 -m mh_z19 --pwm
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/var/lib/python/ansible/lib/python3.12/site-packages/mh_z19/__main__.py", line 104, in <module>
    print (mh_z19.read_from_pwm(gpio=args.pwm_gpio, range=args.pwm_range))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/lib/python/ansible/lib/python3.12/site-packages/mh_z19/__init__.py", line 222, in read_from_pwm
    GPIO.setup(gpio,GPIO.IN)
RuntimeError: Cannot determine SOC peripheral base address
```

UedaTakeyuki, 2024-05-06:
Hi [rewse](https://github.com/rewse), I've updated this to use the gpiozero and uploaded it to the testpypi instead of pypi.

Would you please confirm this and tell me the result? Thank you for your contribution.

[site of testpypi](https://test.pypi.org/project/mh-z19/)
How to update mh-z19 from testpypi

```
pip install --index-url https://test.pypi.org/simple/ mh-z19
```

After it is confirmed, I'll upload this to **pypi**.

UedaTakeyuki, 2024-05-06:
Thank you for your continued cooperation in providing useful information. The gpiozero seems to depend on the lgpio, however, gpiozero's **setup.py** does not have this dependency description.
So, I'll add **lgpio** to mh-z19's **setup.py** in addition to **gpiozero**, and upload to the **pypi**.

UedaTakeyuki, 2024-05-06:
I've uploaded my latest package to the **PyPI** as **version 3.1.5**.

rewse, 2024-05-06:
Confirmed. Thank you!

```
(mhz19-prod) tats@fox:~$ pip3 install mh_z19
(mhz19-prod) tats@fox:~$ pip list
Package            Version
------------------ --------
certifi            2024.2.2
charset-normalizer 3.3.2
colorzero          2.0
getrpimodel        0.1.24
gpiozero           2.0.1
idna               3.7
lgpio              0.2.2.0
mh_z19             3.1.5
pip                24.0
pyserial           3.5
requests           2.31.0
RPi.GPIO           0.7.1
setuptools         69.5.1
urllib3            2.2.1
(mhz19-prod) tats@fox:~$ sudo python3 -m mh_z19 --pwm
{'co2': 485.0}
```

closed 2024-05-06 by rewse

mroelandts, 2025-02-28:
The RPi.GPIO lib dependency is still listed in the pypy/setup.py file.
As it's still imported inside the pypi/mh_z19/pwm.py file. But the function `read_from_pwm_with_gpio` is not used as it has been replaced with `read_from_pwm_with_gpiozero`.
So isn't it better to cleanup and drop the RPi.GPIO dependency?

I'm suggesting this cause I'm currently having install issues on my zero2. (/etc/debian_version -> 12.9 and Python 3.11.2)
And am unable to install rpi-gpio==0.7.1.
```bash
uv pip install mh-z19
Resolved 14 packages in 5.97s
      Built mh-z19==3.1.6
      Built getrpimodel==0.1.26
  × Failed to build `rpi-gpio==0.7.1`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `setuptools.build_meta:__legacy__.build_wheel` failed (exit status: 1)

      [stdout]
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-aarch64-cpython-311/RPi
      copying RPi/__init__.py -> build/lib.linux-aarch64-cpython-311/RPi
      creating build/lib.linux-aarch64-cpython-311/RPi/GPIO
      copying RPi/GPIO/__init__.py -> build/lib.linux-aarch64-cpython-311/RPi/GPIO
      running build_ext
      building 'RPi._GPIO' extension
      creating build/temp.linux-aarch64-cpython-311/source
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC
      -I/home/maki/.cache/uv/builds-v0/.tmp9QJJRH/include -I/usr/include/python3.11 -c source/c_gpio.c -o build/temp.linux-aarch64-cpython-311/source/c_gpio.o
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC
      -I/home/maki/.cache/uv/builds-v0/.tmp9QJJRH/include -I/usr/include/python3.11 -c source/common.c -o build/temp.linux-aarch64-cpython-311/source/common.o

      [stderr]
      source/c_gpio.c: In function ‘setup’:
      source/c_gpio.c:170:9: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
        170 |     if ((uint32_t)gpio_mem % PAGE_SIZE)
            |         ^
      source/c_gpio.c:171:34: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
        171 |         gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
            |                                  ^
      source/common.c:23:10: fatal error: Python.h: No such file or directory
         23 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1

      hint: This error likely indicates that you need to install a library that provides "Python.h" for `rpi-gpio@0.7.1`
  help: `rpi-gpio` (v0.7.1) was included because `mh-z19` (v3.1.6) depends on `rpi-gpio`
```

UedaTakeyuki, 2025-02-28:
Hi [mroelandts](https://github.com/mroelandts), thank you for your report.
I will work to resolve this issue. Please wait a moment.

UedaTakeyuki, 2025-03-02:
Hi [mroelandts](https://github.com/mroelandts),

I've dropped the RPi.GPIO dependency and uploaded it to PyPI as version 3.1.7.
Would you please confirm it? Thank you for your cooperation.

mroelandts, 2025-03-02:
Thanks! Works great.