UedaTakeyuki/mh-z19 #54: Serial Device Must Be Specified When Using Raspberry Pi 5

ziyucao, 2024-08-28:
Hi Takeyuki,
I found that when using Raspberry Pi 5, it is necessary to specify the serial device. Is this expected behavior?

When I connect the MH-Z19E to the Raspberry Pi 5 according to the wiring instructions in the README and run the command
`sudo python3 -m mh_z19` 
the output is 
`{}`

However, when I specify the serial device in this command
`sudo python3 -m mh_z19 --serial_device /dev/ttyAMA0` 
I get an output like 
`{"co2": 1711}`

This might be related to a change in the Raspberry Pi 5.
[https://www.raspberrypi.com/documentation/computers/configuration.html#primary-and-secondary-uart](url)
> `/dev/serial0` and `/dev/serial1` are symbolic links which point to either `/dev/ttyS0` or `/dev/ttyAMA0`.
> On the Raspberry Pi 5, `/dev/serial0` is a symbolic link that points to `/dev/ttyAMA10`.

mh_z19 Version: 3.1.5
Device: MH-Z19E

UedaTakeyuki, 2024-08-28:
Hi ziyucao,

Thank you for your report.
I'll add RPi5 support soon.

UedaTakeyuki, 2024-08-29:
Hi [ziyucao](https://github.com/ziyucao),

I've fixed and uploaded my latest package to the [PyPI](https://pypi.org/project/mh-z19/) as version 3.1.6.
Would you please confirm it? Thank you.

ziyucao, 2024-08-29:
Hi Takeyuki,

It works great on version 3.1.6! Thank you a lot!
[image]

closed 2024-09-02 by UedaTakeyuki