UedaTakeyuki/mh-z19 #55: Some unexpected Error

EliasKramer, 2024-09-19:
# Hi, im rather new to this Raspberry pi thing. So this error could be an obvious fix.
#### I have a Rasp Pi 4 and have only installed the os so far. 
#### after running setup.sh and installing the pip package i ran in the following issue:

```
pi@raspberrypi:~/Desktop/co2reader/mh-219 $ sudo python3 -m mh_z19 
Traceback (most recent call last):

File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 322, in open self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyS0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

File "/home/elias/Desktop/co2reader/mh-z19/mh_z19.py", line 70, in read_concentration ser = connect_serial()
File "/home/elias/Desktop/co2reader/mh-z19/mh_z19.py", line 61, in connect_serial return serial.Serial(serial_dev,
File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 244, in __init____
self.open()
File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 325, in open
raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg)) serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyS0: [Errno 2] No such file or directory: '/dev/ttysS0'
```
The Sensor was connected. I have no idea what I'm doing wrong.

Best Regards

UedaTakeyuki, 2024-09-28:
Hi Elias,

First of all, sorry for my late reply due to my overlooking your issue.
Would you please try the following command and tell me the result? 

```
sudo python3 -m mh_z19 --serial_device /dev/ttyUSB0
```

EliasKramer, 2024-09-28:
Hi, thanks for your time.

```
pi@raspberrypi:~ $ ls /dev
autofs           initctl       pts     tty12  tty39  tty8       vcsu1
block            input         ram0    tty13  tty4   tty9       vcsu2
btrfs-control    kmsg          ram1    tty14  tty40  ttyprintk  vcsu3
bus              kvm           ram10   tty15  tty41  uhid       vcsu4
cachefiles       log           ram11   tty16  tty42  uinput     vcsu5
cec0             loop0         ram12   tty17  tty43  urandom    vcsu6
cec1             loop1         ram13   tty18  tty44  usb        vcsu7
char             loop2         ram14   tty19  tty45  v4l        vga_arbiter
console          loop3         ram15   tty2   tty46  vchiq      vhci
cpu_dma_latency  loop4         ram2    tty20  tty47  vcio       vhost-net
cuse             loop5         ram3    tty21  tty48  vc-mem     vhost-vsock
disk             loop6         ram4    tty22  tty49  vcs        video10
dma_heap         loop7         ram5    tty23  tty5   vcs1       video11
dri              loop-control  ram6    tty24  tty50  vcs2       video12
fb0              mapper        ram7    tty25  tty51  vcs3       video13
fd               media0        ram8    tty26  tty52  vcs4       video14
full             media1        ram9    tty27  tty53  vcs5       video15
fuse             media2        random  tty28  tty54  vcs6       video16
gpiochip0        media3        rfkill  tty29  tty55  vcs7       video18
gpiochip1        mem           shm     tty3   tty56  vcsa       video19
gpiomem          mmcblk0       snd     tty30  tty57  vcsa1      video20
hidraw0          mmcblk0p1     stderr  tty31  tty58  vcsa2      video21
hidraw1          mmcblk0p2     stdin   tty32  tty59  vcsa3      video22
hidraw2          mqueue        stdout  tty33  tty6   vcsa4      video23
hidraw3          net           tty     tty34  tty60  vcsa5      video31
hidraw4          null          tty0    tty35  tty61  vcsa6      watchdog
hwrng            port          tty1    tty36  tty62  vcsa7      watchdog0
i2c-20           ppp           tty10   tty37  tty63  vcsm-cma   zero
i2c-21           ptmx          tty11   tty38  tty7   vcsu
```

I realize i do not have ttyUsb0 and i do not have /dev/serial

The OS is btw Debian 12

UedaTakeyuki, 2024-09-29:
Umm, no UART.

May I ask you to show me your ***config*** as follows?
```cat /boot/config.txt```

Also, ***gpio*** status as:
```raspi-gpio get```

UedaTakeyuki, 2024-10-13:
3 weeks no feedback.

closed 2024-10-13 by UedaTakeyuki